antmicro/gerber2ems #1: Installation issue (needs clarification) tested with Ubuntu22.04 LTS

bvernoux, 2023-11-28:
I have tried to do the installation step by step on Ubuntu 22.04 LTS
All is fine for first steps
But when we arrive on the step "Script installation" https://github.com/antmicro/gerber2ems#script-installation
It is not clear if before we shall be in the `gerber2ems` root directory as if it is the case you shall add the step to clone it and cd to the right directory 
- `git clone https://github.com/antmicro/gerber2ems.git`

Anyway when we do `pip install .` in `gerber2ems ` (with Python Virtual Environment set before to be changed as it is done after) directory there is strange output

```
(.venv) bvern@Ben:~/gerber2ems$ pip install .
Processing /home/bvern/gerber2ems
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: UNKNOWN
  Building wheel for UNKNOWN (pyproject.toml) ... done
  Created wheel for UNKNOWN: filename=UNKNOWN-0.0.0-py3-none-any.whl size=5111 sha256=85209c45fe3446ae3dbd4faa57349b7dedbfb4cd49b869967e8392aed0cf5aae
  Stored in directory: /tmp/pip-ephem-wheel-cache-ocdpq2fa/wheels/91/10/db/b11266faa64a38de400911a2c79d0ce159ecc8257d6526e636
Successfully built UNKNOWN
Installing collected packages: UNKNOWN
  Attempting uninstall: UNKNOWN
    Found existing installation: UNKNOWN 0.0.0
    Uninstalling UNKNOWN-0.0.0:
      Successfully uninstalled UNKNOWN-0.0.0
Successfully installed UNKNOWN-0.0.0
```
Also thanks to clarify the remaining step after as it is not clear where gerber2ems.sh is intended to be ...

picli3, 2023-11-30:
I also have the same problem on Linux Mint. I assume that the repo should be cloned and the command executed like this
```
git clone https://github.com/antmicro/gerber2ems.git
cd gerber2ems 
pip install .
```
look what it shows me.

[image]

I will be waiting in case there is new news.

That's all for today!

pkatarzynski, 2023-12-01:
You may need to update pip before proceeding with installation. 
We have added an additional step to  https://github.com/antmicro/gerber2ems#script-installation.

bvernoux, 2023-12-02:
https://github.com/antmicro/gerber2ems/blob/main/README.md has been updated this issue is closed

closed 2023-12-02 by bvernoux