antmicro/gerber2ems #2: not fully consistent documentation, samples

oe2lsp, 2023-11-29:
Hi,
thanks for the project!

I'm trying one of your samples (basic).
In the README the format of the CSV for  the ports uses spaces as delimiter, but in the code (importer.py function: get_ports_from_file) "," is used as delimiter.

If someone is new to all of this, at least one example with all necessary files available would be great to get started.

I tried to generate the geometry and view the generated xml in AppCSXCAD but nothing looks correct. The top layer with the trace is missing, no vias and the position from the ports I read from an gerber-viewer are way off.
So I assume its my mistake, but without a working example getting started is quite a challenge :)

pkatarzynski, 2023-12-01:
Hi, 

We have added some samples to https://github.com/antmicro/gerber2ems/tree/main/examples for reference.

bvernoux, 2023-12-01:
It is not very clear how to use paraview (https://github.com/antmicro/gerber2ems#paraview) with example provided especially as when I have built both example basic & differential (gerber2ems -a) in simulation output there is lot of files generated as output in ems/simulation/... without any extension and documentation (https://github.com/antmicro/gerber2ems#paraview) speak about ems/simulation/<port_number>/e_field_<layer_number>_..vtr so we expect .vtr extension ?
Does I have missed something ?

oe2lsp, 2023-12-03:
Thanks,
all examples and a test project of my own are working fine.

With the given examples one question rises. In job-simulation-part there are some lumped components defined as ports, is there a parsing of these parameters planed for the future?

kevinjuszczyk, 2024-01-07:
> ems/simulation/<port_number>/e_field_<layer_number>_..vtr so we expect .vtr extension ?

@bvernoux, vtr files are produced using the --export-field option
`gerber2ems -a --export-field`

DLentka, 2025-04-11:
Hi,

We updated README, installation section should be more straightforward to follow.

The examples were also updated - now they contain slices of our OSHW [Signal Integrity Test Board](https://openhardware.antmicro.com/boards/si-simulation-test-board/?view=top-ortho&tab=features), generated using the [KiCad SI wrapper](https://github.com/antmicro/kicad-si-simulation-wrapper). Selected examples contain VNA measurement results (`vna.csv`).

closed 2025-04-22 by wgryncewicz