antmicro/gerber2ems #9: add_ports() causes output files to be opened twice

jwise, 2024-04-21:
I discovered when running a badly set up simulation that the simulator seems to open all of the output files twice, and as a result, if it is not internally consistent, it will write bizarre output files.  They look as follows, and the postprocessing stage chokes on them.  (I apologize for the screenshot rather than pasted output... I fixed my simulation and am not sure how I originally triggered this issue :-) )

[image]

When I debugged this, I noticed that we are `Adding ports` twice, and indeed, when I instrumented openEMS, I found that multiple `ProcessIntegral::InitProcess` was being called on the same filename multiple times.  It looks like what's happening is that the `add_ports(sim)` in the `geometry` step is saving each of these ports to the `geometry.xml`, but then when we go `add_ports(sim, excite)` later after after we've already reloaded the `geometry.xml`, then we cause the ports to be instantiated into the simulator twice.

I think the solution is that `geometry` should not `add_ports`, but I am not entirely sure I understand openEMS well enough to say exactly whether this is true or not!  But doing this does, at least, fix that `InitProcess` is called only once per filename.

jwise, 2024-04-21:
Ah, no, my proposed fix completely breaks the simulation :)

But it is still the case I think that having `add_ports` twice goes a little strangely.

DLentka, 2025-04-11:
Hi @jwise,
Adding ports in geometry step is intended, as it allows to examine port placement in AppCSXCAD.
Recently We have updated gerber2ems and now simulation step is able to reuse ports that are already defined in geometry.xml instead re-adding them.

closed 2025-04-22 by wgryncewicz