antmicro/gerber2ems #17: differential pair results look wrong

aholtzma-am, 2025-01-08:
Using commit 387977f2c9c1d81b32dfaa8f5f574cd85069fac9 the differential pair results don't look right:
[image]

Here is the log:

```
10:17:36][INFO] Parsing config
[10:17:36][WARN] No field ['name'] found in config. Using default: Unnamed
[10:17:36][WARN] No field ['dB_margin'] found in config. Using default: -15
[10:17:36][WARN] No field ['name'] found in config. Using default: Unnamed
[10:17:36][WARN] No field ['dB_margin'] found in config. Using default: -15
[10:17:36][WARN] No field ['name'] found in config. Using default: Unnamed
[10:17:36][WARN] No field ['dB_margin'] found in config. Using default: -15
[10:17:36][WARN] No field ['name'] found in config. Using default: Unnamed
[10:17:36][WARN] No field ['dB_margin'] found in config. Using default: -15
[10:17:36][WARN] No field ['traces'] found in config. Using default: []
[10:17:36][INFO] Creating geometry
[10:17:36][INFO] Processing gerber files
[10:17:36][INFO] Adding copper from gerber files
[10:17:36][INFO] Processing gerber files
[10:17:36][INFO] Adding contours for F_Cu
[10:17:36][INFO] Adding contours for B_Cu
[10:17:36][INFO] Mesh line count, x: 102, y: 222 z: 65. Total number of cells: ~1.47M
[10:17:36][INFO] Adding substrates
[10:17:36][INFO] Adding MUR boundary condition
[10:17:36][INFO] Adding vias from excellon file
[10:17:36][INFO] Adding ports
[10:17:36][INFO] Saving geometry to /Users/aholtzma/src/gerber2ems/examples/differential/ems/geometry/geometry.xml
[10:17:36][INFO] Running simulation
[10:17:36][INFO] Simulating with excitation on port #0
[10:17:36][INFO] Loading geometry from /Users/aholtzma/src/gerber2ems/examples/differential/ems/geometry/geometry.xml
[10:17:36][INFO] Adding ports
[10:17:36][WARN] Port #-1 is defined twice on the board. Ignoring the second instance
[10:17:36][WARN] Port #0 is defined twice on the board. Ignoring the second instance
[10:17:36][WARN] Port #1 is defined twice on the board. Ignoring the second instance
[10:17:36][WARN] Port #2 is defined twice on the board. Ignoring the second instance
[10:17:36][INFO] Starting simulation
 ----------------------------------------------------------------------
 | openEMS 64bit -- version v0.0.36-62-g078483d
 | (C) 2010-2023 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ----------------------------------------------------------------------
	Used external libraries:
		CSXCAD -- Version: v0.6.3-26-g9ab4e88
		hdf5   -- Version: 1.14.5
		          compiled against: HDF5 library version: 1.14.5
		tinyxml -- compiled against: 2.6.2
		fparser
		boost  -- compiled against: 1_76
		vtk -- Version: 9.4.1
		       compiled against: 9.4.1

Create FDTD operator (compressed SSE + multi-threading)
FDTD simulation size: 106x222x65 --> 1.52958e+06 FDTD cells
FDTD timestep is: 1.43828e-13 s; Nyquist rate: 173 timesteps @2.00946e+10 Hz
Excitation signal length is: 2214 timesteps (3.18436e-10s)
Max. number of timesteps: 10000 ( --> 4.51671 * Excitation signal length)
Create FDTD engine (compressed SSE + multi-threading)
Warning: No primitives found in property: Plane!
Warning: No primitives found in property: Via!
Warning: No primitives found in property: ViaFilling!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
[... 77 lines cut ...]
Warning: Unused primitive (type: Polygon) detected in property: Gerber_0!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_1!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_1!
Warning: Unused primitive (type: Polygon) detected in property: Gerber_1!
Running FDTD engine... this may take a while... grab a cup of coffee?!?
[@        4s] Timestep:          602 || Speed:  218.4 MC/s (7.002e-03 s/TS) || Energy: ~6.34e-18 (- 0.00dB)
[@        8s] Timestep:         1720 || Speed:  413.9 MC/s (3.695e-03 s/TS) || Energy: ~1.13e-15 (- 0.00dB)
[@       12s] Timestep:         3268 || Speed:  586.3 MC/s (2.609e-03 s/TS) || Energy: ~1.99e-17 (-17.53dB)
[@       16s] Timestep:         5246 || Speed:  750.1 MC/s (2.039e-03 s/TS) || Energy: ~1.86e-17 (-17.82dB)
[@       20s] Timestep:         7568 || Speed:  882.5 MC/s (1.733e-03 s/TS) || Energy: ~1.77e-17 (-18.05dB)
RunFDTD: Warning: Max. number of timesteps was reached before the end-criteria of -60dB was reached...
	You may want to choose a higher number of max. timesteps...
Time for 10000 iterations with 1529580.00 cells : 24.16 sec
Speed: 633.01 MCells/s
[10:18:27][INFO] Postprocessing
[10:18:27][INFO] Adding virtual ports
[10:18:28][INFO] Processing all data from simulation. Calculating S-parameters and impedance
[10:18:28][INFO] Rendering S-parameter plots
[10:18:28][INFO] Rendering impedance plots
[10:18:28][INFO] Rendering smith charts
[10:18:28][INFO] Rendering differential pair S-parameter plots
[10:18:28][INFO] Rendering differential pair impedance plots
[10:18:28][INFO] Rendering trace delay plot
```
I tried increasing the max time steps to 1e5, but that didn't help appreciably.

aholtzma-am, 2025-01-08:
Is there a known good commit that I can run here to see if the problems is in gerber2ems or openems?

DLentka, 2025-04-11:
Hi @aholtzma-am,

We have updated examples in our repository with slices of our open hardware [Signal Integrity Test Board](https://openhardware.antmicro.com/boards/si-simulation-test-board/?view=top-left&tab=features. This board allows us to reference openEMS simulation results to VNA measurements.

Our other open source designs which contain differential traces simulated using `gerber2ems`, include:
* [GMSL Deserializer Board](https://openhardware.antmicro.com/boards/gmsl-deserializer/?view=top-ortho&tab=simulation&si-track-id=%2FConnectors%2FCSI1.CLK_PN&tab-fullscreen=true)
* [Environment Sensor MYRA SiP Baseboard](https://openhardware.antmicro.com/boards/environment-sensor-sip-baseboard/?view=top-ortho&tab=simulation&si-track-id=%2FUSB_MCU_PN&tab-fullscreen=true)
* [Composite Video (CVBS) to MIPI CSI-2 bridge](https://openhardware.antmicro.com/boards/cvbs-mipi-bridge/?view=top-ortho&tab=simulation&tab-fullscreen=true&si-track-id=%2FCSI_A_CLK_PN)

I'm attaching [Environment Sensor MYRA SiP Baseboard](https://github.com/antmicro/environment-sensor-myra-baseboard) `USB_MCU_Diff` differential impedance graph for reference.

[image]

We've also updated our README to include recommended openEMS-project commit.

closed 2025-04-22 by wgryncewicz