antmicro/gerber2ems #18: ModuleNotFoundError: No module named 'CSXCAD'

soypat, 2025-01-20:
After installing all dependencies and running pip install succesfully I'm getting the following error. I'm using uv package manager.

```
✗ uv run python3 src/gerber2ems/main.py "$@"
warning: No `requires-python` value found in the workspace. Defaulting to `>=3.9`.
Traceback (most recent call last):
  File "/home/pato/src/gerber2ems/src/gerber2ems/main.py", line 16, in <module>
    from gerber2ems.simulation import Simulation
  File "/home/pato/src/gerber2ems/src/gerber2ems/simulation.py", line 9, in <module>
    import CSXCAD
ModuleNotFoundError: No module named 'CSXCAD'
```

soypat, 2025-01-21:
I believe the following commands helped me get further along the install process
```
uv pip install setuptools
uv pip install cython # Or maybe sudo apt install cython3?
uv pip install --no-build-isolation  $PATH_TO_OPENEMS_REPO/CSXCAD/python
```

But now I have the following issue:
```
✗ uv pip install --no-build-isolation ../openEMS-Project/CSXCAD/python
Resolved 1 package in 5.80s
  × Failed to build `csxcad @ file:///home/pato/src/openEMS-Project/CSXCAD/python`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `setuptools.build_meta:__legacy__.build_wheel` failed (exit status: 1)

      [stdout]
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/SmoothMeshLines.py -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/__init__.py -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/CSTransform.pxd -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/CSXCAD.pxd -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/ParameterObjects.pxd -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/CSProperties.pxd -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/CSRectGrid.pxd -> build/lib.linux-x86_64-cpython-39/CSXCAD
      copying CSXCAD/CSPrimitives.pxd -> build/lib.linux-x86_64-cpython-39/CSXCAD
      running build_ext
      building 'CSXCAD.CSPrimitives' extension
      creating build/temp.linux-x86_64-cpython-39/home/pato/src/openEMS-Project/CSXCAD/python/CSXCAD
      g++ -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -DOPENSSL_NO_SSL3 -fPIC
      -I/home/pato/src/gerber2ems/.venv/include -I/home/pato/.pyenv/versions/3.9.18/include/python3.9
      -c /home/pato/src/openEMS-Project/CSXCAD/python/CSXCAD/CSPrimitives.cpp -o
      build/temp.linux-x86_64-cpython-39/home/pato/src/openEMS-Project/CSXCAD/python/CSXCAD/CSPrimitives.o

      [stderr]
      /home/pato/src/openEMS-Project/CSXCAD/python/CSXCAD/CSPrimitives.cpp:1258:10: fatal error:
      CSXCAD/ParameterObjects.h: No such file or directory
       1258 | #include "CSXCAD/ParameterObjects.h"
            |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/g++' failed with exit code 1

      hint: This error likely indicates that you need to install a library that provides
      "CSXCAD/ParameterObjects.h" for `csxcad @ file:///home/pato/src/openEMS-Project/CSXCAD/python`
```

pkatarzynski, 2025-01-23:
Hi @soypat,

The problem seems to be caused by `uv` as it by default creates a venv that does not inherit global Python packages (and as CSXCAD is getting installed during installation of [OpenEMS-project](https://github.com/thliebig/openEMS-Project.git) it is probably installed as such).

The solution is to force creation of venv with `--system-site-packages` flag (similar to what we currently propose for manual venv creation in [Readme](https://github.com/antmicro/gerber2ems?tab=readme-ov-file#virtual-environment)), using `uv` this can be achieved by `uv venv --system-site-packages` after this command (provided that OpenEMS was installed correctly) `uv run [...]` should be able to find CSXCAD.

closed 2025-01-28 by pkatarzynski