antmicro/gerber2ems #20: RecursionError: maximum recursion depth exceeded

taogya, 2025-03-08:
The following commit gives an error:
https://github.com/antmicro/gerber2ems/commit/1fc3d5becbd2e33b6e605d1d96310da8e5772c5f

Error Detail:
```
% cd examples/differential
% gerber2ems -a
Parsing config
Loading config from /path/to/your/gerber2ems/examples/differential/simulation.json
[13:29:22][INFO] Creating geometry
[13:29:22][INFO] Processing gerber files (may take a while for larger boards)
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.11/3.11.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
           ^^^^^^^^^^^^^^^^
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/importer.py", line 66, in gbr_to_png
    dpi = 1 / (cfg.pixel_size * UNIT / 0.0254)
               ^^^^^^^^^^^^^^
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/config.py", line 238, in __getattr__
    return getattr(self._config, name)
                   ^^^^^^^^^^^^
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/config.py", line 238, in __getattr__
    return getattr(self._config, name)
                   ^^^^^^^^^^^^
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/config.py", line 238, in __getattr__
    return getattr(self._config, name)
                   ^^^^^^^^^^^^
  [Previous line repeated 2989 more times]
RecursionError: maximum recursion depth exceeded
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/path/to/your/venv/bin/gerber2ems", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/main.py", line 49, in main
    geometry(sim)
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/main.py", line 82, in geometry
    importer.process_gbrs_to_pngs()
  File "/path/to/your/venv/lib/python3.11/site-packages/gerber2ems/importer.py", line 54, in process_gbrs_to_pngs
    p.map(partial(gbr_to_png, edge), layers)
  File "/opt/homebrew/Cellar/python@3.11/3.11.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.11/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/pool.py", line 774, in get
    raise self._value
RecursionError: maximum recursion depth exceeded
```

This is probably because the contents of Confi.load are not passed to child processes in multi-processing.
Below is an example of a fix, which worked for me.
```
# gerber2ems/src/gerber2ems/main.py

from gerber2ems.config import Config, parse_arguments

def main() -> None:
    """Run the script."""
    args = parse_arguments()
    # move to config.py vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
    # ### Config.load(args)
    # ### setup_logging(args)
    # move to config.py ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    if args.update_config:
        exit(0)
...
# move to config.py vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
# ### def parse_arguments() -> argparse.Namespace
...
# ### def setup_logging(args: Any) -> None:
...
# move to config.py ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

```
# gerber2ems/src/gerber2ems/config.py

import argparse
import coloredlogs
...
def parse_arguments() -> argparse.Namespace:
    """Parse commandline arguments."""
    parser = argparse.ArgumentParser(
        prog="EM-Simulator",
        description="This application allows to perform EM simulations based on standard PCB production files (gerber)",
    )
    parser.add_argument(
        "-c", "--config", metavar="CONFIG_FILE", help="Path to config file [default: `./simulation.json`]"
    )
    parser.add_argument("--update-config", action="store_true", help="Add missing fields to config file")
    parser.add_argument("-g", "--geometry", action="store_true", help="Create geometry")
    parser.add_argument("-s", "--simulate", action="store_true", help="Run simulation")
    parser.add_argument("-p", "--postprocess", action="store_true", help="Postprocess the data")
    parser.add_argument(
        "-a", "--all", action="store_true", help="Execute all steps (geometry, simulation, postprocessing)"
    )
    parser.add_argument(
        "--export-field", "--ef", action="store_true", help="Export electric field data from the simulation"
    )

    group = parser.add_mutually_exclusive_group()
    group.add_argument("-d", "--debug", action="store_true", dest="debug")
    group.add_argument("-l", "--log", choices=["DEBUG", "INFO", "WARNING", "ERROR"], dest="log_level")

    return parser.parse_args()

def setup_logging(args: Any) -> None:
    """Set up logging based on command line arguments."""
    level = logging.INFO
    if args.debug:
        level = logging.DEBUG
    if args.log_level is not None:
        level = logging.getLevelName(args.log_level)

    if level == logging.DEBUG:
        coloredlogs.install(
            fmt="[%(asctime)s][%(name)s:%(lineno)d][%(levelname).4s] %(message)s",
            datefmt="%H:%M:%S",
            level=level,
            logger=logger,
        )
    else:
        coloredlogs.install(
            fmt="[%(asctime)s][%(levelname).4s] %(message)s",
            datefmt="%H:%M:%S",
            level=level,
            logger=logger,
        )

    to_disable = ["PIL", "matplotlib"]
    for name in to_disable:
        disabled_logger = logging.getLogger(name)
        disabled_logger.setLevel(logging.ERROR)

args = parse_arguments()
Config.load(args)
setup_logging(args)
# ### end of file
```

taogya, 2025-03-08:
The latest commit (https://github.com/antmicro/gerber2ems/commit/1fc3d5becbd2e33b6e605d1d96310da8e5772c5f) is several times faster in my environment, so it's very good.

taogya, 2025-04-07:
Hi.
I ran [latest commit](https://github.com/antmicro/gerber2ems/commit/428346f1d1ef7e5c2bc67ee252246b9a1a1f0c30).
This issue is resolved.

Thank you for your support !!

closed 2025-04-07 by taogya

DLentka, 2025-04-11:
Hi @tagoya,
The issue was caused by difference in default method used for creating tasks in multiprocessing.Pool. On Linux (platform we primarily use for testing) Python uses fork method, while on MacOS & Windows spawn is used (with fork, child inherits variables, with spawn not). As you have already noticed, we have pushed an update that will ensure that no matter which method is used, Config will be passed to worker threads.