antmicro/gerber2ems #23: Paraview: Python Environment Management

stellarpower, 2025-05-23:
Hi,

I've managed to run the stub_short example simulation successfully. When I run `ems2paraview 0` on the output, Paraview opens, but nothing is shown but the default start screen, and I get an error to the console that it's failed to import the paraview module.

I do see the same if I open paraview as a standalone, so, it presumably is related to my system's paraview installation - but it seems that Paraview is bundled with its own python installation (available as `pvpython`), so, I am just wondering what the intended way that gerber2ems and paraview should be interfacing each other is.

Should I be running the `ems2paraview` script using Paraview's interpreter? Or with the same interpreter where I installed `gerber2ems`? Should the package be installed to Paraview's environment too, or should it not make a difference?

Will try stepping through tomorrow, but thought better to confirm what the correct setup is rather than assume.

Thanks.

stellarpower, 2025-05-23:
If anyone else is wondering, it's as simple as e.g. 
`GERBER2EMS_PORT=0   /explicit/path/to/paraview  --script   /usr/local/lib/python3.11/site-packages/ems2paraview/paraview_preview.py`

I had more success with a download of Paraview 6 RC 1 as a tarball, as opposed to the version in my package manager, for whatever reason. The original script wraps a subprocess, so you can simply emplace paraview in your path, and presumably the preview is then the one running in the embedded environment.

DLentka, 2025-06-17:
Hi @stellarpower,

It seems that this is not an issue with `gerber2ems`/`ems2paraview` but with Paraview installation from your package manager.
Similar case: https://discourse.paraview.org/t/modulenotfounderror-no-module-named-paraview/10301).  

Consider trying following approach: uninstall `paraview` from your package manager, install binary from https://www.paraview.org/download/, add it to PATH, and simply use `ems2paraview <port num>`.

stellarpower, 2025-06-26:
Thanks, but

> Should I be running the ems2paraview script using Paraview's interpreter? Or with the same interpreter where I installed gerber2ems? Should the package be installed to Paraview's environment too, or should it not make a difference?

what I'm asking is it'd still help to have some clarification on whether ems2paraview is meant to be running in the same environment, a different one, if I should be using pvpython, etc. - rather than asking for a solution to the message I am getting from within paraview. From the above, it seems that `paraview_preview.py` isn't depending on much so it can just be run in paraview's internal environment without installing the package, but, I'm not sure if this is how it was meant to be used.

Also I've forgotten about what I was doing at the time since opening this issue, but, am sure I tried what you suggested with several paraview builds. From memory, I think I did have to sidestep the `paraview2ems` script and explicitly hook it in through paraview's, as then gerber2ems is not installed within their pvpython.

Thanks